HKUST-Aerial-Robotics/VINS-Mono #451: How should I handle a picture of the vehicle itself taken? Crop or resize?

SerpentWindy, 2024-05-20:
Due to angle limitations, my camera may capture the vehicle body. Do I need to crop or resize the photos? What should I do?

SerpentWindy, 2024-05-20:
As shown in the following figure:
[image]

SerpentWindy, 2024-05-22:
The turning angle of the deployed VINS algorithm on the vehicle's trajectory is basically consistent with the actual trajectory, while the distance between the VINS trajectory and the map trajectory is significantly different.

How should i do?

The VINS trajectory diagram is shown below：

[image]

The map trajectory is shown below
[image]
：

brucezhcw, 2025-05-13:
i think it's easy to make a mask to cover car body,  make sure feature point will not on car body