HKUST-Aerial-Robotics/VINS-Mono #452: How to align the trajectory with google map?

Echo-gh, 2024-05-23:
I need to align camera trajectory from ORB-SLAM2 with google map. I've found that there are similar results in  the Vins-mono paper. However, the authors didn't mention about how to make it. Anyone can help me?
What I've got:

- ORB-SLAM2 camera trajectory file (transformation matrix)
- The original position in ECEF obtained from GPS.
- The original attitude (Euler angles)

SerpentWindy, 2024-05-24:
May I ask if you have tried taking screenshots of Google Maps and camera trajectories separately, and then overlaying the two images together?

Echo-gh, 2024-05-24:
> May I ask if you have tried taking screenshots of Google Maps and camera trajectories separately, and then overlaying the two images together?

No, I don't think this method can work well. What I want is to get an accurate result.

SerpentWindy, 2024-05-24:
> > May I ask if you have tried taking screenshots of Google Maps and camera trajectories separately, and then overlaying the two images together?
> 
> No, I don't think this method can work well. What I want is to get an accurate result.

I'm really sorry, but I don't know what to do.

Echo-gh, 2024-05-24:
> > > May I ask if you have tried taking screenshots of Google Maps and camera trajectories separately, and then overlaying the two images together?
> > 
> > 
> > No, I don't think this method can work well. What I want is to get an accurate result.
> 
> I'm really sorry, but I don't know what to do.

Not at all. Thanks for replying!

closed 2024-05-28 by Echo-gh