HKUST-Aerial-Robotics/VINS-Mono #462: Increased drift when vehicle is stationary

yeoeun98, 2024-12-13:
Hello,
I test VINS-Mono.

Drift increases when vehicle is stationary.

[image]
[image]

I don't know why this happens, and I don't know how to debug it.

< about setting >
( CAM )
img hz : 10hz
img size : 1920 x 1080

(IMU)
imu hz : 125hz

brucezhcw, 2025-03-02:
in stationary state, scale factor is not observable. maybe scale problem

if stationary case is usually in your test case, you can add ZUPT module in VINS

vladniculescu, 2025-09-30:
How would you implement the ZUPT module? In case low parallax is detected, how would you constrain the optimization problem?

brucezhcw, 2025-09-30:
The static state can be detected by the parallax of the visual image or the variance observed by IMU, and can also be used in combination. 

In the static state, it is necessary to greatly weaken or even cancel the IMU pre-integration constraint at the back end, and also cancel the initial value provided by IMU integration

means VIO should be VO